fubarlabs/foocars #89: Consider adjusting the logic for car_runner.py support autonomous mode and data collection at the same time.

rianders, 2021-05-08:
Consider adjusting the logic for car_runner.py to support autonomous mode and data collection at the same time. The existing code appears to be capable of support that. It would also make the button logic easier.

ricklon, 2021-08-15:
This is a must-have. When trying to compare how different models are working we need to have recordings and data to compare different autonomous runs.

ricklon, 2021-08-15:
Error when both collect and autonomous happen simultaneously.

```
Data Collection: On
Autonomous: On
Traceback (most recent call last):
  File "/foocars/cars/carservices/carservices/carRunner.py", line 200, in callback_switch_autonomous
    g_camera.start_recording(g_getter, format='rgb')
  File "/usr/local/lib/python3.9/site-packages/picamera/camera.py", line 1043, in start_recording
    camera_port, output_port = self._get_ports(True, splitter_port)
  File "/usr/local/lib/python3.9/site-packages/picamera/camera.py", line 561, in _get_ports
    raise PiCameraAlreadyRecording(
picamera.exc.PiCameraAlreadyRecording: The camera is already using port 1
0.09378170967102051
Autonomous: Off
0.05613398551940918
Data Collection: Off
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/picamera/camera.py", line 1189, in stop_recording
    encoder = self._encoders[splitter_port]
KeyError: 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/foocars/cars/carservices/carservices/carRunner.py", line 238, in callback_switch_collect_data
    g_camera.stop_recording()
  File "/usr/local/lib/python3.9/site-packages/picamera/camera.py", line 1191, in stop_recording
    raise PiCameraNotRecording(
picamera.exc.PiCameraNotRecording: There is no recording in progress on port 1
Data Collection: Off
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/picamera/camera.py", line 1189, in stop_recording
    encoder = self._encoders[splitter_port]
KeyError: 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/foocars/cars/carservices/carservices/carRunner.py", line 238, in callback_switch_collect_data
    g_camera.stop_recording()
  File "/usr/local/lib/python3.9/site-packages/picamera/camera.py", line 1191, in stop_recording
    raise PiCameraNotRecording(
picamera.exc.PiCameraNotRecording: There is no recording in progress on port 1
```

ricklon, 2021-08-15:
This is a conflict when both activities want to use the camera device at the same time.