fubarlabs/foocars #114: Add car modes to the carservices.

ricklon, 2021-08-14:
Add car modes to the current defines.py file.  The modes would be:
* manual, Manual mode is how the car operates normally.
* auto, Tells the car to start in autonomous mode.
* remote, Tell the car to start in web remote control mode.

ricklon, 2021-08-14:
Adding car modes but chasing errors:
```

0.3430147171020508
0.3357412815093994
[2.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 734504.0, 1500.0, 1500.0]
0.342418909072876
Traceback (most recent call last):
  File "/foocars/cars/carservices/carservices/carRunner.py", line 207, in callback_switch_autonomous
    if callback_switch_autonomous.is_auto==True or auto_mode==False:
NameError: name 'auto_mode' is not defined
```

rianders, 2021-08-15:
Another catch is when is the behavior of the switch. False positives can happen and false lows because of "bounce"  So the autonomous callback checks to see if it's called more than once.  Here's what the function does:

* 1st Time HIGH turn on the camera and start imagesetter sending to Tensorflow
* 2nd or more times HIGH: ignore
* 1st time LOW shutdown camera and imagegetter
* 2nd time LOW ignore

What happens when the define MODE=auto occurs is it gets stuck in the "ignore" the extra highs.

ricklon, 2021-08-15:
Car modes are added to the defines and the carRunner.py

closed 2021-08-15 by ricklon