fubarlabs/foocars #183: RPi.GPIO library version update to 0.7.1 needed for 64 bit Raspberry PI

ricklon, 2023-05-01:
Error using 0.7.0 and 64 bit Raspberry PI. Tested using 0.7.1 works.

```
  /usr/bin/ld: build/temp.linux-aarch64-cpython-39/source/soft_pwm.o:/tmp/tmpk6txtnh9/RPi.GPIO-0.7.0/source/soft_pwm.c:28: multiple definition of `threads'; build/temp.linux-aarch64-cpython-39/source/event_gpio.o:/tmp/tmpk6txtnh9/RPi.GPIO-0.7.0/source/event_gpio.c:60: first defined here
  collect2: error: ld returned 1 exit status
  error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1

  at /usr/local/lib/python3.9/dist-packages/poetry/installation/chef.py:152 in _prepare
      148│
      149│                 error = ChefBuildError("\n\n".join(message_parts))
      150│
      151│             if error is not None:
    → 152│                 raise error from None
      153│
      154│             return path
      155│
      156│     def _prepare_sdist(self, archive: Path, destination: Path | None = None) -> Path:

Note: This error originates from the build backend, and is likely not a problem with poetry but with rpi.gpio (0.7.0) not supporting PEP 517 builds. You can verify this by running 'pip wheel --use-pep517 "rpi-gpio (==0.7.0)"'.
```

rianders, 2023-05-19:
Consider moving to libgpiod
https://lloydrochester.com/post/hardware/libgpiod-intro-rpi/

rianders, 2023-05-19:
Heres the python library:
https://pypi.org/project/gpiod/